ilg-archived/openocd #2: OpenOCD silent installations on windows 7

rgkandur, 2015-09-29:
I am trying to install https://github.com/gnuarmeclipse/openocd/releases/download/gae-0.9.0-20150519/gnuarmeclipse-openocd-win64-0.9.0-201505190955-setup.exe from command line. I am successfully able to install silently with /S option but failed to install to different folder than default. The silent installations always installing openocd at "C:\Program Files\GNU ARM Eclipse\OpenOCD". Kindly let me know how install to different folder than default one.

Following are the commands i used to install openocd at "L:\openocd":

C:>"C:\Users\rkandur\Downloads\gnuarmeclipse-openocd-win64-0.9.0-201505190955-setup.exe" /S /D=l:\openocd

ilg-ul, 2015-09-29:
I currently have no experience with Windows silent installations, so I'll need some time to investigate.

ilg-ul, 2015-09-29:
According to the NSIS manual, it should work:

> /D sets the default installation directory ($INSTDIR), overriding InstallDir and InstallDirRegKey. It must be the last parameter used in the command line and must not contain any quotes, even if the path contains spaces. Only absolute paths are supported.

The NSIS script used to build the setup is publicly available from: [gnuarmeclipse/openocd](https://github.com/gnuarmeclipse/openocd/blob/gnuarmeclipse-dev/gnuarmeclipse/nsis/gnuarmeclipse-openocd.nsi). 

Unfortunately I have limited Windows testing capabilities, so if you spot any problems that would prevent /D to work properly, please suggest a patch and I can try to fix it.

ilg-ul, 2016-02-17:
any progress? can we close this ticket?

ilg-ul, 2016-10-29:
fixed in v0.10.0-201610281609-dev

closed 2016-10-29 by ilg-ul