ilg-archived/openocd #3: undefined symbols (dyld)

postspectacular, 2016-01-22:
Hi @ilg-ul, I've just tried your pre-build version of OCD to get support for STM32F746-DISCO, but getting dynamic linker errors (on OSX 10.8.5). Previously I've been running the "official" OCD 0.9.0 without problems...

```
/Applications/GNU\ ARM\ Eclipse/OpenOCD/0.10.0-201601101000-dev/bin/openocd -f board/stm32f7discovery.cfg
GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-dev-00287-g85cec24 (2016-01-10-12:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
dyld: lazy symbol binding failed: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Applications/GNU ARM Eclipse/OpenOCD/0.10.0-201601101000-dev/bin/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

dyld: Symbol not found: _OSAtomicIncrement32Barrier
  Referenced from: /Applications/GNU ARM Eclipse/OpenOCD/0.10.0-201601101000-dev/bin/libusb-1.0.0.dylib
  Expected in: /usr/lib/libSystem.B.dylib

Trace/BPT trap: 5
```

cnoviello, 2016-02-09:
I've the same problem here

ilg-ul, 2016-02-09:
> 10.8.5

hmmm... the build environment is 10.11.x, and I take special precautions to pack all required libraries, but  the system libraries obviously differ.

for such old versions I'm afraid you have to rebuild the binary, I doubt there is any reasonable solution to use binaries built on recent versions.

closed 2016-02-17 by ilg-ul