ilg-archived/openocd #4: Unable to set breakpoints by line number under Windows (symptom: "No source named C:\\\\")

dsbeach, 2016-01-24:
OpenOCD versions tested: 0.9 and 0.10
Eclipse version: Mars

Problem:
Project builds successfully and loads on the chip OK.  Initial breakpoint is successful. Attempting to set new breakpoints via source window fail as shown below:
```
758,755 &"No source file named C:\\\\Development\\\\Workspace\\\\DiscoveryF0\\\\Src\\\\main.c.\n"
758,757 52^done,bkpt={number="5",type="breakpoint",disp="keep",enabled="y",addr="<PENDING>",pending=\
"C:\\\\Development\\\\Workspace\\\\DiscoveryF0\\\\Src\\\\main.c:76",times="0",original-location="C:\\
\\\Development\\\\Workspace\\\\DiscoveryF0\\\\Src\\\\main.c:76"}
758,758 (gdb) 
```

However, setting breakpoints via disassembly window are successful.
```
617,172 55-break-insert --thread-group i1 -f *0x080001de
617,249 55^done,bkpt={number="7",type="breakpoint",disp="keep",enabled="y",addr="0x080001de",func="S\
ystemClock_Config",file="../Src/main.c",fullname="C:\\Development\\Workspace\\DiscoveryF0\\Debug/../\
Src/main.c",line="105",times="0",original-location="*0x080001de"}
```

I am working to see if I can resolve this issue myself. However, in doing so I ran into an issue with the build process on ubuntu.  The build script failed with references to /Host/Work/....

I've resolved that issue and it looks like I can now build.  Would you like the build problem entered as a separate issue here?

dsbeach, 2016-01-24:
Here are my notes on getting the build to work.  I tried several things but I think what cured it was the following:

copy ~/Work to /Home/Work
cd ~/Work/openocd/gnuarmeclipse-openocd.git
make distclean
bash ~/Downloads/build-openocd.sh --all

dsbeach, 2016-01-24:
Looks like this may be more of an issue in the eclipse plugin.  Moving my investigation there, but would like to leave this open for now.

dsbeach, 2016-01-24:
Just a status update, I'm done for today.

I have cloned the plugin repository, built the plugin, and successfully stepped through some plugin code.  I think I'm on the right track, but assistance as to where to concentrate my efforts would be appreciated.  Hopefully I'll have some more time tomorrow.

ilg-ul, 2016-01-25:
I'm not sure I understand your problem.

setting and managing breakpoints is not done in the openocd plug-in, but in cdt.

the openocd plug-in just starts/stops the executable and provides a gui to configure it.

dsbeach, 2016-01-25:
Thanks, I'm starting to understand the issue more clearly.

Apparently there are a host of issues related to backslash handling of Windows paths under gdb debugging.  At this point I may just break down a get a native unix machine for this project.

I'd like to find a way to 'coerce' the elf symbols into forward slashes, but I don't know if that is possible.

I'm fine with closing this issue.  However the problems I had doing a build on ubuntu should be addressed.  Should that be opened as a new problem?  In this repository, or some other?

Thanks again,
David

ilg-ul, 2016-01-25:
> the problems I had doing a build on ubuntu should be addressed. Should that be opened as a new problem? In this repository, or some other?

since it is related to the build scripts, I suggest to use that repo.

please clearly explain the problem, so I can understand and reproduce it, and suggest a solution.

dsbeach, 2016-01-25:
Thanks.  I'll close this issue and open a new issue in the build scripts repo in the next few days.

I was able to come up with a suitable work-around for this problem, so I'm finally in a position to debug my own project today.

The work-around involves overriding the debugging source defaults.  First in the "Debug Configurations..." remove the default path and add "Path Relative to Source Folders".

[image]

Then in "Project Properties" ensure all the paths you want to be able to add breakpoints to are specified.

[image]

Thanks again for your response to this issue, as well as all the work you and others have put forward to make OpenOCD such a useful tool.

David

closed 2016-01-25 by dsbeach

ilg-ul, 2016-01-25:
David,

could you post your findings in the project forum?

https://www.element14.com/community/groups/gnu-arm-eclipse