ilg-archived/openocd #5: uninstaller removes entire directory in V0.10~

mmprestine, 2016-03-01:
Caution!!!!

uninstaller removes entire directory in V0.10~ on windows install.

ilg-ul, 2016-03-01:
that's correct, the uninstaller removes the entire install folder, but it does this **after** the user confirms the operation.

if you don't want the folder removed, just click **Cancel** and the uninstall will be terminated.

mmprestine, 2016-03-03:
The problem here is that if the user installs under an alternate directory that they have created just for development then it uninstalls everything in that directory and not just the software.  To me that is a huge problem.

ilg-ul, 2016-03-03:
sorry for any inconveniences.

however I'm not sure I understand what you suggest. if you want to preserve the openocd folder, just don't run the uninstall.

I can add some warnings in the documentation pages, but I doubt they'll make any difference.

closed 2016-10-29 by ilg-ul