ilg-archived/openocd #8: Need additional step for linux installation (http://gnuarmeclipse.github.io/openocd/install/)

sledgeh, 2016-03-07:
I followed installation [manual](http://gnuarmeclipse.github.io/openocd/install/) on Ubuntu 15.10 and met an issue:

     libusb_open failed: LIBUSB_ERROR_ACCESS

No error was with `sudo`. The solution is to add user to `plugdev` group. 

     usermod -aG plugdev $USER

Than relogin or restart. FIXED.

ilg-ul, 2016-03-07:
yes, thank you for the info, it looks like each different Linux version has its own way of dealing with usb access rights :-(

I'll update the documentation.

ilg-ul, 2016-10-29:
the explanation with `plugdev` was already present in the Troubleshooting section at the end of the page. to avoid further confusion, I moved it to the GNU/Linux section.

closed 2016-10-29 by ilg-ul