ilg-archived/openocd #9: STM32F4 Discovery init mode failed

tlamb96, 2016-05-27:
When I try to connect via the command:
sudo ./openocd -f board/stm32f4discovery.cfg
I get the following output:
http://pastebin.com/dcPBQ52Q
I have googled and cannot find an answer. I have the 49-stlinkv2.rules, 99-openocd.rules, and stlink.conf in my /etc/udev/rules.d folder. I have the most recent firmware on my board from here:
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link004.html
Do you know what's going wrong?

ilg-ul, 2016-05-27:
for general support please use the project forum.

tlamb96, 2016-06-07:
My STM32F4 board was shipped with Read Out Protection to level 1 which disables any writing to the memory. As an STM32 novice, this took quite a lot of time to figure out but is now fixed.

closed 2016-06-07 by tlamb96

ilg-ul, 2016-06-07:
ok, thank you for the info.