ilg-archived/openocd #10: New STM32F72x MCU's are not yet supported

MJ666, 2017-02-27:
Working with an STM Nucleo STM32F722 board for debugging with Eclipse and Windows. This board is just available since a few days. I get the following output:

	GNU ARM Eclipse 64-bits Open On-Chip Debugger 0.10.0-00113-g0f83948 (2017-01-24-18:48)
	Licensed under GNU GPL v2
	For bug reports, read
		http://openocd.org/doc/doxygen/bugs.html
	Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
	adapter speed: 2000 kHz
	adapter_nsrst_delay: 100
	srst_only separate srst_nogate srst_open_drain connect_deassert_srst
	srst_only separate srst_nogate srst_open_drain connect_deassert_srst
	Started by GNU ARM Eclipse
	Info : Unable to match requested speed 2000 kHz, using 1800 kHz
	Info : Unable to match requested speed 2000 kHz, using 1800 kHz
	Info : clock speed 1800 kHz
	Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
	Info : using stlink api v2
	Info : Target voltage: 3.247244
	Info : stm32f7x.cpu: hardware has 8 breakpoints, 4 watchpoints
	Info : accepting 'gdb' connection on tcp/3333
	Info : device id = 0x10006452
	Warn : Cannot identify target as a STM32 family.
	Error: auto_probe failed
	Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
	Error: attempted 'gdb' connection rejected

I think I know what needs to be added to src/flash/nor/ stm32f2x.c file. Setting up an complete build environment looks to be a bit complicated. Im working on windows. Is there are someone who has the environment to make an windows 64 bit test build for me with the required updates to the file?

https://github.com/gnuarmeclipse/openocd/blob/gnuarmeclipse-dev/src/flash/nor/stm32f2x.c#L66
 * STM32F7[2|3]
 * 512 KByte part with 4 x 16, 1 x 64, 3 x 128.
 *

https://github.com/gnuarmeclipse/openocd/blob/gnuarmeclipse-dev/src/flash/nor/stm32f2x.c#L895
	case 0x452:	/* F72x/73x */
		max_flash_size_in_kb = 512;
		max_sector_size_in_kb = 128;
		flash_size_reg = 0x1FF0F442;       //not sure for this one, but likely the same as other F7 ???
		stm32x_info->has_extra_options = true;
		stm32x_info->has_boot_addr = true;
		break;

https://github.com/gnuarmeclipse/openocd/blob/gnuarmeclipse-dev/src/flash/nor/stm32f2x.c#L1137
	case 0x452:
		device_str = "STM32F7[2|3]x";

		switch (rev_id) {
		case 0x1000:
			rev_str = "A";
			break;
		}
		break;

ilg-ul, 2017-02-27:
the procedure to generate the win64/win32 binaries runs on linux, with a docker image.

if you recreate the build environment on windows, with msys2 and mingw-w64, I guess you can also build openocd, then experiment with any changes you need.

but finally the changes should be passed to the OpenOCD project, the GNU ARM Eclipse OpenOCD fork just repacks the programs in a more convenient binary package.

MJ666, 2017-03-01:
I got this to work with using msys2 with the patches above. Unfortunately the resulting OpenOCD looks to have a complete different structure and is not very portable. I will try to get the updates to the OpenOCD project.

closed 2017-03-01 by MJ666

ilg-ul, 2017-03-01:
> a complete different structure and is not very portable

well, I never tried a build on windows, but if you follow the same steps as in the current build script, it should produce the same result.

> I will try to get the updates to the OpenOCD project

yes, that's better, but it might take a while.

MJ666, 2017-03-01:
I used the procedure described here for building: http://www.playembedded.org/blog/en/2015/10/11/building-openocd-under-windows-using-msys2/

ilg-ul, 2017-03-01:
how about this:

http://gnuarmeclipse.github.io/openocd/build-procedure/

MJ666, 2017-03-04:
I have already seen this but its only for building on Mac or Linux. Would be an overkill to install an Linux only for this. At the moment ist working fro me. I send the OpenOCD guys an e-mail with the updates.

ilg-ul, 2017-03-04:
> Would be an overkill ...

yes, but based on the details in my script, you can write your own simple script, specific for your environment.

zhiyb, 2017-06-23:
From STM32F72xxx reference manual (section 35.2), flash size register might be
```flash_size_reg = 0x1FF07A22;```

MJ666, 2017-06-25:
Could debug with my custom build version. Likely will wait for an official support. There is an pending PR already fro OpenOCD. Many thanks anyhow.