ilg-archived/openocd #13: host system call

Daparrag, 2017-05-01:
Dear all, 

I'm working with the STM32F401RE board in the semihosting mode, However was impossible to execute system call's related with the command processor (shell(LINUX) or the command line(WINDOWS)) such as "dir, ls -l", I suppose that it is because Openocd doesn't support such kind of SYSTEM_CALLS.

If my affirmation is correct, I'm Interesting in implementing such functionalities, could you bring me some support or ideas of how I can complete that task, please.  

I will appreciate your help so much. 

bye

ilg-ul, 2017-05-01:
I did not check the OpenOCD sources, but I also remember some problems with the newlib implementation of the `system()` call.

I suggest you check both the OpenOCD and newlib sources.

Daparrag, 2017-05-01:
could you please specified which are exactly the files in  OpenOCD where is included the implementation for the system calls.

thanks you

ilg-ul, 2017-05-01:
> which are exactly the files...

sure. aaaa... nope. top secret! ;-)

Daparrag, 2017-05-01:
thanks for your answers. I will start to work on that.

closed 2017-05-01 by Daparrag

ilg-ul, 2017-05-01:
good luck. be sure you fix newlib first.