ilg-archived/openocd #17: RISC-V target support doesn't build without --disable-werror

TM1234, 2017-06-27:
Again not a major issue but a small annoying build issue.

I realise that you build using Docker but I am building on MinGW-64/MSYS2 using the i686 (32 bit environment) as follows:

```
git clone --recursive https://github.com/gnu-mcu-eclipse/openocd openocd-gme
cd openocd-gme
./bootstrap
./configure --prefix=`pwd`/openocd
make 
```
However the build fails due to warnings (treated as errors by default) in the RISC-V target support:
```
In file included from ./config.h:340:0,
                 from src/target/riscv/riscv-011.c:11:
src/target/riscv/riscv-011.c: In function 'read_memory':
./src/helper/replacements.h:30:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                         ^
src/target/riscv/riscv-011.c:2303:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 3 - i, max_batch_size);
                             ^
./src/helper/replacements.h:30:38: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                                      ^
src/target/riscv/riscv-011.c:2303:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 3 - i, max_batch_size);
                             ^
src/target/riscv/riscv-011.c: In function 'write_memory':
./src/helper/replacements.h:30:25: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                         ^
src/target/riscv/riscv-011.c:2464:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 2 - i, max_batch_size);
                             ^
./src/helper/replacements.h:30:38: error: signed and unsigned type in conditional expression [-Werror=sign-compare]
 #define MIN(a, b) (((a) < (b)) ? (a) : (b))
                                      ^
src/target/riscv/riscv-011.c:2464:29: note: in expansion of macro 'MIN'
   unsigned int batch_size = MIN(count + 2 - i, max_batch_size);
                             ^
cc1.exe: all warnings being treated as errors
make[2]: *** [Makefile:2900: src/target/riscv/riscv-011.lo] Error 1
make[2]: Leaving directory '/home/Tommy.Murphy/dev/openocd/test/openocd-gme'
make[1]: *** [Makefile:3797: all-recursive] Error 1
make[1]: Leaving directory '/home/Tommy.Murphy/dev/openocd/test/openocd-gme'
make: *** [Makefile:1630: all] Error 2
```
To build successfully I have to use:
```
./configure --prefix=`pwd`/openocd --disable-werror [--enable-riscv]
```
Note that building the official OpenOCD 0.10.0+dev sources without specifying --disable-werror at configuration time works fine:
```
git clone --recursive https://git.code.sf.net/p/openocd/code openocd-official
cd openocd-official
./bootstrap
./configure --prefix=`pwd`/openocd 
make 
make install
```

ilg-ul, 2017-06-27:
> Docker ...  MSYS2 ... 

for quick development cycles you can use whatever environment you prefer, but for public releases a controlled build in a controlled environment is recommended.

> the build fails due to warnings

I know, I silence these annoying warnings in my scripts. the purpose of the GNU MCU Eclipse OpenOCD fork is mainly to facilitate the integration into the GNU MCU ecosystem, I do not have the resources to fix errors/warnings, so please address these suggestions to the RISC-V maintainers.

> RISC-V target support files get compiled even if --disable-riscv is not specified

I noticed this, but I do not know the dependencies of this file. in the QEMU build there are many conditional definitions in the makefile fragments, perhaps something similar can be used here.

please feel free to suggest imprevements in the form of pull requests.

TM1234, 2017-06-27:
I've logged the issue against the SiFive openocd fork here:

https://github.com/riscv/riscv-openocd/issues/71

So I'll close this one - I presume that's OK?

closed 2017-06-27 by TM1234

ilg-ul, 2017-06-27:
ok