ilg-archived/openocd #18: "Remote failure reply: E0E" when OpenOCD is piped

ivankravets, 2017-08-10:
Hi @ilg-ul,

I don't know where is better to post this issue, would be thankful for a help. I can't reproduce this issue with `20170607-2132-dev` but with `v0.10.0-2-20170622-1535` I can.

Just pipe OpenOCD to GDB:

```
(gdb) target remote | openocd -c "gdb_port pipe" ...
...
undefined debug reason 7 - target needs reset
Remote failure reply: E0E
```

ilg-ul, 2017-08-11:
I'll investigate.

TM1234, 2017-09-02:
This may be due to the GDB/OpenOCD startup/init "race condition" whereby GDB gets started and tries to make a GDB Remote Serial Protocol connection before OpenOCD is quite ready to handle it? This can happen with the socket/port connection but maybe the same issue can arise with the pipe/stdio connection too? See here for more on this issue when using the socket/port connection:

https://github.com/gnu-mcu-eclipse/eclipse-plugins/issues/186

If you can launch OpenOCD standalone and then get GDB to connect to it OK then that may be evidence in support of this race condition being the problem.

Hope this helps.

ilg-ul, 2017-09-02:
I know there were some changes in OpenCOD that I did not study thoroughly, so it might be so.

as far as I know, there are no changes in GNU MCU Eclipse OpenOCD compared to master related to this issue, and if ther are, they probably come from merging the RISC-V code.

JojoS62, 2017-09-03:
I get this error message also after exporting a project from mbed. Target is a nrf51822, debug probe a LPC-LINK2 with cmsis-dap firmware. Calling oocd 0.10.0-3-20170826-1813-dev manually for flashing works, but starting to debug gives the e0e error code. Also when I try to start gdb manually and to connect with 'target remote localhost:3333'.
When I use oocd 0.10.0 compiled from F. Chopin the debugger starts without errors.

ilg-ul, 2017-09-25:
looks the same as https://github.com/riscv/riscv-openocd/issues/115

ilg-ul, 2018-01-14:
can you confirm if this was fixed in the latest binary release?

ivankravets, 2018-01-15:
Re-tested. It works with the latest GNU MCU Eclipse OpenOCD v0.10.0-6-20180112

Thanks and Happy New Year!

closed 2018-01-15 by ivankravets

ilg-ul, 2018-01-15:
you're welcome, happy new year!