ilg-archived/openocd #20: Readme.md - "How to build" link is broken

TM1234, 2017-08-31:
Hi Liviu - small issue with the readme.md here - the "How to build" link is broken.

I guess that http://gnuarmeclipse.github.io/openocd/how-to-build should be (or should redirect to?) https://gnu-mcu-eclipse.github.io/openocd/build-procedure/.

Also should the "Eclipse plug-in" text be a link to https://gnu-mcu-eclipse.github.io/debug/openocd/ ?

Regards
Tommy

ilg-ul, 2017-08-31:
Thank you, Tommy, you are right, the links were broken.

I updated them, please check.

TM1234, 2017-08-31:
The "How to build" link now works for me - thanks.
The "Eclipse plug-in" text is still not a link and I'm not sure if it should be?
All other links worked fine before and still work (just in case they were changed).
Thanks for the quick update.

ilg-ul, 2017-08-31:
> The "Eclipse plug-in" text is still not a link ...

oops! 

fixed.

TM1234, 2017-08-31:
It is now. :-) Thanks. 
I guess it's OK for me to close this now?

closed 2017-08-31 by TM1234

ilg-ul, 2017-08-31:
> I guess it's OK for me to close this now?

sure, that's the recommended procedure.

if necessary, I can always reopen it.