ilg-archived/openocd #24: Broken submodule links to repo.or.cz

jakob223, 2018-04-03:
### Description

Submodule links are broken to [repo.or.cz](repo.or.cz). They are of the form `repo.or.cz/r/something.git`, but the actual URLs seem to be `repo.or.cz/something.git`

### Steps to Reproduce

1. try to update submodules

**Expected behaviour:** Submodules are cloned

**Actual behaviour:** Failure to connect to download them

```
[~/workspace/openocd]$ git submodule update --init                                                                                                                                                                  [gnu-mcu-eclipse-dev]
Cloning into '/Users/jakobw/workspace/openocd/jimtcl'...
fatal: unable to access 'http://repo.or.cz/r/jimtcl.git/': Couldn't connect to server
fatal: clone of 'http://repo.or.cz/r/jimtcl.git' into submodule path '/Users/jakobw/workspace/openocd/jimtcl' failed
Failed to clone 'jimtcl'. Retry scheduled
Cloning into '/Users/jakobw/workspace/openocd/src/jtag/drivers/libjaylink'...
fatal: unable to access 'http://repo.or.cz/r/libjaylink.git/': Couldn't connect to server
fatal: clone of 'http://repo.or.cz/r/libjaylink.git' into submodule path '/Users/jakobw/workspace/openocd/src/jtag/drivers/libjaylink' failed
Failed to clone 'src/jtag/drivers/libjaylink'. Retry scheduled
Cloning into '/Users/jakobw/workspace/openocd/tools/git2cl'...
fatal: unable to access 'http://repo.or.cz/r/git2cl.git/': Couldn't connect to server
fatal: clone of 'http://repo.or.cz/r/git2cl.git' into submodule path '/Users/jakobw/workspace/openocd/tools/git2cl' failed
Failed to clone 'tools/git2cl'. Retry scheduled
Cloning into '/Users/jakobw/workspace/openocd/jimtcl'...
fatal: unable to access 'http://repo.or.cz/r/jimtcl.git/': Couldn't connect to server
fatal: clone of 'http://repo.or.cz/r/jimtcl.git' into submodule path '/Users/jakobw/workspace/openocd/jimtcl' failed
Failed to clone 'jimtcl' a second time, aborting
```

ilg-ul, 2018-04-12:
Thank you for your report. I plan to merge the latest changes from the master repo, but I do not have a date for this.

ilg-ul, 2018-05-13:
I just released 0.10.0-8, with the latest commits. 

Could you check if the problem was solved?