ilg-archived/openocd #25: Unexpected "Remote connection closed" with GDB PIPE and RISC-V

ivankravets, 2018-04-03:
We had a similar problem with ARM before which was resolved a few months ago (https://github.com/gnu-mcu-eclipse/openocd/issues/18). The similar issue could be reproduced for RISC-V now:

```
(gdb) target remote | openocd -c "gdb_port pipe" -f openocd/scripts/board/sifive-hifive1.cfg
Remote debugging using | openocd -c "gdb_port pipe" -f openocd/scripts/board/sifive-hifive1.cfg
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-16:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x10e31913 (mfg: 0x489 (SiFive, Inc.), part: 0x0e31, ver: 0x1)
Info : [0] Found 2 triggers
halted at 0x204001e2 due to debug interrupt
Info : Examined RISCV core; XLEN=32, misa=0x40001105
Info : Found flash device 'issi is25lp128' (ID 0x0018609d)
cleared protection for sectors 64 through 255 on flash bank 0
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection from pipe
Info : dropped 'gdb' connection
Remote connection closed
(gdb)
```

/cc @palmer-dabbelt

ilg-ul, 2018-04-03:
I have no idea what was wrong, I'm only tracking both the main repo and the risc-v repo.

I'll try to update, but I'd rather wait for the risc-v fork to be upstreamed, there are too many differences to accomodate.

ivankravets, 2018-04-03:
I tried RISC-V fork. The same problem. 
What is interesting, it works good on Linux.

ilg-ul, 2018-04-03:
> I tried RISC-V fork. The same problem.

then open a ticket there.

ilg-ul, 2018-05-13:
I just released 0.10.0-8, with the latest commits. There are many cleanups in the RISC-V code.

Could you check if the problem was solved?

ivankravets, 2018-06-05:
Thanks for the update. I've just tested and still can reproduce this issue.

ilg-ul, 2018-10-21:
could you retry with the latest release (*-10)?

maskimthedog, 2018-11-06:
This issue is kills my ability to use Code Confidence plugin.

maskimthedog, 2018-11-06:
...on Windows 10 that is with System Workbench and STLink V2.

ilg-ul, 2018-11-06:
I suggest you repeat the same test using the Windows binaries packed by Freddy Chopin and let me know the result.

I am aware of some problems introduced by the RISC-V patches in previous releases, but the latest `-10` release should be clean, I removed most of the RISC-V patches in the common code.

ilg-ul, 2019-04-29:
I just released 0.10.0-12, with the latest upstream master.

Please check if the problem was solved. If not, reopen the ticket.

closed 2019-04-29 by ilg-ul

maskimthedog, 2019-04-29:
Thank you.  I will test.