ilg-archived/openocd #27: double free or corruption with ST-LINK/V2

j123b567, 2018-04-16:
### Description

I have board without hardware reset and when I accidentaly configure `reset_config srst_only`, openocd failes with "double free or corruption".

I can connect with ST-LINK/V2-1 without problem.
(ST-LINK/V2-1 is reported as `STLINK v2 JTAG v30 API v2 SWIM v20 VID 0x0483 PID 0x374B`)

After connecting with ST-LINK/V2, OpenOCD fails with backtrace and double free or corruption message.
(ST-LINK/V2 is reported as `STLINK v2 JTAG v30 API v2 SWIM v7 VID 0x0483 PID 0x3748`)

I have tried all versions of gnu-mcu-eclipse/openocd and the problem was not in version
`v0.10.0-2-20170622-1535-dev`
and it first occures in
`v0.10.0-3-20170826-1813-dev`
all later versions are affected, at least
`v0.10.0-7-20180123`
`v0.10.0-5-20171110`

### Steps to Reproduce

1. Connect ST-Link/V2 (not V2-1) to target MCU using just GND, TVCC, SWDIO, SWCLK
2. Run OpenOCD with this configuration (I'm using STM32F030RC). It is wrong configuration, because SRST is not connected, so it should just tell "timed out while waiting for target halted".

```tcl
source [find interface/stlink-v2.cfg ]
reset_config srst_only
set CHIPNAME STM32F030xC
set WORKAREASIZE 0x00008000
source [find target/stm32f0x.cfg ]
tcl_port disabled
telnet_port disabled
gdb_port disabled
init
reset halt
sleep 300
flash write_image erase "app.elf"
sleep 300
reset run
sleep 1000
shutdown
```

**Expected behaviour:** [What you expected to happen]

```
Error: timed out while waiting for target halted
TARGET: STM32F030xC.cpu - Not halted
```

**Actual behaviour:** [What actually happened]

```
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00404-g20463c28 (2018-01-23-12:24)
...
Error: timed out while waiting for target halted
Error in `/path_to/bin/openocd': double free or corruption (fasttop): 0x000000000287e2c0 
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f05389e97e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f05389f237a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f05389f653c]
/opt/scide/cnd/tools/Linux-x86_64/openocd/0.10.0-7-20180123-1217/bin/openocd[0x5df57f]
```

### Versions

* OpenOCD package version  v0.10.0-3-20170826-1813-dev and later
* Debug probe: ST-Link/V2 (VID 0x0483 PID 0x3748)
* operating system: Various x86_64 linux distributions (Ubuntu 16.04, Gentoo)

ilg-ul, 2018-04-16:
thank you for reporting this, but I don't maintain openocd, I just pack it and distribute for user convenience.

I suggest you try the latest openocd version built from the main repository, and, if the problem is still there, report it to the maintainers.

j123b567, 2018-04-16:
You are producing heavily patched version of OpenOCD. I can't reproduce this with upstream OpenOCD.
Upstream version 0.10.0 works
Upstream version from 20180117 (commit edb67962865d5d3cc4a8ec1790b4c8c5327e98fd) works
Upstream version head (commit 2b47ded8dee5ba52d47ce7cbc8b643ae4a3663ae) works

ilg-ul, 2018-04-16:
Oops! It might be possible that some of the RISC-V patches to interfere with ST-LINK.

The RISC-V guys are cleaning their code and contributing upstream. 

I'll try to make a new release, hopefully it'll be less problematic.

ilg-ul, 2018-05-13:
I just released 0.10.0-8, with the latest commits. There are many cleanups in the RISC-V code.

Could you check if the problem was solved?

faisal-shah, 2018-06-14:
I'm still getting a similar problem ... I ran OpenOCD in gdb, and also printed out the backtrace. Hope that helps ..

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Open On-Chip Debugger 0.10.0+dev-dirty (2018-06-01-19:51)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 500 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : clock speed 480 kHz
[New Thread 0x7ffff6ec2700 (LWP 3558)]
Info : STLINK v2 JTAG v28 API v2 SWIM v18 VID 0x0483 PID 0x374B
Info : using stlink api v2
Info : Target voltage: 3.225000
Info : stm32l4x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
target halted due to debug-request, current mode: Handler External Interrupt(86)
xPSR: 0x21000066 pc: 0x08004a40 msp: 0x200003f0
Info : device id = 0x20006461
Info : flash size = 1024kbytes
Info : Unable to match requested speed 500 kHz, using 480 kHz
Info : Unable to match requested speed 500 kHz, using 480 kHz
adapter speed: 480 kHz
Error: timed out while waiting for target halted
*** Error in `/usr/local/bin/openocd': double free or corruption (fasttop): 0x0000000000967490 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7ffff735f7e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7ffff736837a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7ffff736c53c]
/usr/local/bin/openocd[0x556416]
======= Memory map: ========
00400000-00667000 r-xp 00000000 fd:00 263489                             /usr/local/bin/openocd
00867000-00868000 r--p 00267000 fd:00 263489                             /usr/local/bin/openocd
00868000-00872000 rw-p 00268000 fd:00 263489                             /usr/local/bin/openocd
00872000-00b01000 rw-p 00000000 00:00 0                                  [heap]
7ffff0000000-7ffff0021000 rw-p 00000000 00:00 0 
7ffff0021000-7ffff4000000 ---p 00000000 00:00 0 
7ffff64ac000-7ffff64c2000 r-xp 00000000 fd:00 787560                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff64c2000-7ffff66c1000 ---p 00016000 fd:00 787560                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff66c1000-7ffff66c2000 rw-p 00015000 fd:00 787560                     /lib/x86_64-linux-gnu/libgcc_s.so.1
7ffff66c2000-7ffff66c3000 ---p 00000000 00:00 0 
7ffff66c3000-7ffff6ec3000 rw-p 00000000 00:00 0 
7ffff6ec3000-7ffff6eca000 r-xp 00000000 fd:00 827899                     /lib/x86_64-linux-gnu/librt-2.23.so
7ffff6eca000-7ffff70c9000 ---p 00007000 fd:00 827899                     /lib/x86_64-linux-gnu/librt-2.23.so
7ffff70c9000-7ffff70ca000 r--p 00006000 fd:00 827899                     /lib/x86_64-linux-gnu/librt-2.23.so
7ffff70ca000-7ffff70cb000 rw-p 00007000 fd:00 827899                     /lib/x86_64-linux-gnu/librt-2.23.so
7ffff70cb000-7ffff70e3000 r-xp 00000000 fd:00 817470                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ffff70e3000-7ffff72e2000 ---p 00018000 fd:00 817470                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ffff72e2000-7ffff72e3000 r--p 00017000 fd:00 817470                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ffff72e3000-7ffff72e4000 rw-p 00018000 fd:00 817470                     /lib/x86_64-linux-gnu/libpthread-2.23.so
7ffff72e4000-7ffff72e8000 rw-p 00000000 00:00 0 
7ffff72e8000-7ffff74a8000 r-xp 00000000 fd:00 817662                     /lib/x86_64-linux-gnu/libc-2.23.so
7ffff74a8000-7ffff76a8000 ---p 001c0000 fd:00 817662                     /lib/x86_64-linux-gnu/libc-2.23.so
7ffff76a8000-7ffff76ac000 r--p 001c0000 fd:00 817662                     /lib/x86_64-linux-gnu/libc-2.23.so
7ffff76ac000-7ffff76ae000 rw-p 001c4000 fd:00 817662                     /lib/x86_64-linux-gnu/libc-2.23.so
7ffff76ae000-7ffff76b2000 rw-p 00000000 00:00 0 
7ffff76b2000-7ffff76b5000 r-xp 00000000 fd:00 817836                     /lib/x86_64-linux-gnu/libdl-2.23.so
7ffff76b5000-7ffff78b4000 ---p 00003000 fd:00 817836                     /lib/x86_64-linux-gnu/libdl-2.23.so
7ffff78b4000-7ffff78b5000 r--p 00002000 fd:00 817836                     /lib/x86_64-linux-gnu/libdl-2.23.so
7ffff78b5000-7ffff78b6000 rw-p 00003000 fd:00 817836                     /lib/x86_64-linux-gnu/libdl-2.23.so
7ffff78b6000-7ffff79be000 r-xp 00000000 fd:00 782001                     /lib/x86_64-linux-gnu/libm-2.23.so
[... 90 lines cut ...]
#64 0x000000000056bed8 in Jim_IfCoreCommand (interp=0x8e6050, argc=5, argv=0x7fffffffd2e0) at jim.c:11978
#65 0x000000000055fec8 in JimInvokeCommand (interp=interp@entry=0x8e6050, objc=5, objv=objv@entry=0x7fffffffd2e0) at jim.c:10114
#66 0x0000000000560a2c in Jim_EvalObj (interp=interp@entry=0x8e6050, scriptObjPtr=0x9167b0) at jim.c:10565
#67 0x000000000056bf2c in Jim_IfCoreCommand (interp=0x8e6050, argc=5, argv=0x7fffffffd440) at jim.c:11989
#68 0x000000000055fec8 in JimInvokeCommand (interp=interp@entry=0x8e6050, objc=5, objv=objv@entry=0x7fffffffd440) at jim.c:10114
[... cut ...]
#84 0x00000000004511d2 in server_loop (command_context=command_context@entry=0x8e6010) at src/server/server.c:533
#85 0x0000000000405b9c in openocd_thread (cmd_ctx=0x8e6010, argv=0x7fffffffde48, argc=<optimized out>) at src/openocd.c:341
#86 openocd_main (argc=<optimized out>, argv=0x7fffffffde48) at src/openocd.c:378
#87 0x00007ffff7308830 in __libc_start_main (main=0x405470 <main>, argc=5, argv=0x7fffffffde48, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffde38) at ../csu/libc-start.c:291
#88 0x0000000000405569 in _start ()

ilg-ul, 2018-06-14:
> Open On-Chip Debugger 0.10.0+dev-dirty (2018-06-01-19:51)

where did you get this version?

bureaua, 2019-01-17:
I can confirm that this issue is present by using 
gnu-mcu-eclipse-openocd-0.10.0-10-20181020-0522-centos64.tgz

ilg-ul, 2019-01-18:
I just published a new release; I updated to the latest master and removed all RISC-V patches.

Can you test it and confirm that the problem was solved?

ilg-ul, 2019-01-18:
> It looks like it works now

Great, thank you!

@j123b567 Jan, can you also confirm that the problem was solved, and close the ticket?

rafacouto, 2019-01-30:
I am suffuring this behaviour in "0.10.0+dev-00392-gbe9ef0b0", the current version provided by platformio.org. 

@ilg-ul Which tag or commit have you applied the patch?

ilg-ul, 2019-01-30:
> Which tag or commit have you applied the patch?

in the latest release I removed all RISC-V patches, it is mostly the master repacked.

the details of the builds can be seen in:

https://github.com/gnu-mcu-eclipse/openocd-build/blob/master/scripts/container-build.sh

rafacouto, 2019-01-30:
False alarm: uC is damaged and OpenOCD binary dumps, but this is another story... Ignore my comment since it is not related to this issue... Thanks anyway :-)

ilg-ul, 2019-04-29:
I just released 0.10.0-12, with the latest upstream master.

Please check if the problem was solved. If not, reopen the ticket.

closed 2019-04-29 by ilg-ul