ilg-archived/openocd #35: Incompatible date of binaries for macOS

ivankravets, 2018-10-19:
Hi,

I've just downloaded the latest release `0.10.0-9-20181016-1725` for macOS. However, inside all files are dated by July 24, 2018. Is this OK?

Also, I see that new version does not understand double braces `program {{$SOURCE}}`.

Thanks!

ilg-ul, 2018-10-19:
>  inside all files are dated by July 24, 2018. Is this OK?

oops! no, it is not ok, I'll rebuild this package.

> I see that new version does not understand double braces `program {{$SOURCE}}`

could you please elaborate?

ivankravets, 2018-10-19:
> could you please elaborate?

Please ping me here when a new version will be available. Maybe, this issue is already fixed.

ilg-ul, 2018-10-20:
please check 

https://github.com/gnu-mcu-eclipse/openocd/releases/tag/v0.10.0-10-20181020

ilg-ul, 2018-10-20:
ah, tcl... I'm afraid I can't be of much help.

openocd uses jimtcl as a submodule, and I checked, my latest release points to a relatively recent commit of jimtcl (Aug 18).

I suggest you report the problem to the jimtcl project.

closed 2018-10-20 by ilg-ul