ilg-archived/openocd #36: 20181020-553 build non-functional with STM32 F7/H7

TessHull, 2018-12-02:
### Description

When attempting to connect to an F7 nucleo or an H7 nucleo, an invalid-command message is received.
Command in use:
`/opt/gnu-mcu-eclipse/openocd/0.10.0-10-20181020-0522/bin/openocd -f board/st_nucleo_h743zi.cfg`
Result:
```
GNU MCU Eclipse 64-bit Open On-Chip Debugger 0.10.0+dev-00352-gaa6c7e9b (2018-10-20-06:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1800 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
/opt/gnu-mcu-eclipse/openocd/0.10.0-10-20181020-0522/bin/..//scripts/target/stm32h7x.cfg:102: Error: invalid command name "stm32h7x.dap"
in procedure 'script' 
at file "embedded:startup.tcl", line 60
at file "/opt/gnu-mcu-eclipse/openocd/0.10.0-10-20181020-0522/bin/..//scripts/board/st_nucleo_h743zi.cfg", line 8
at file "/opt/gnu-mcu-eclipse/openocd/0.10.0-10-20181020-0522/bin/..//scripts/target/stm32h7x_dual_bank.cfg", line 2
at file "/opt/gnu-mcu-eclipse/openocd/0.10.0-10-20181020-0522/bin/..//scripts/target/stm32h7x.cfg", line 102
```
(The F7 attempt was in the office, I don't have that board to hand here.)

OpenOcd support suggests that this is a mis-match between the scripts and the binary.

I have produced my own openocd build from the upstream source, which works for me. However, I do not wish to become a package maintainer, so I am feeding this issue back to you.

ilg-ul, 2018-12-02:
thank you, I'll try to address this with the next release. 

would simply updating to upstream be enough? should I check something special?

TessHull, 2018-12-02:
I'm afraid I can't answer those questions(lack of knowledge on my part), though I'm happy to try builds against the H7 board on my desk.

All I can confirm is that 42f1cc576ab9b503fadd0b8916a139cd0bc6563e from https://repo.or.cz/openocd.git definitely works with the H7 nucleo.

ilg-ul, 2018-12-02:
ok, I'll check the differences to this commit.

ilg-ul, 2019-04-29:
I just released 0.10.0-12, with the latest upstream master.

Please check if the problem was solved. If not, reopen the ticket.

closed 2019-04-29 by ilg-ul