ilg-archived/openocd #37: Upstream SAMD51 support merged to master

Timvrakas, 2019-02-07:
https://github.com/platformio/platform-atmelsam/issues/43
http://openocd.zylin.com/gitweb?p=openocd.git;a=commit;h=deaf3d264123391d8fe5c4cccbf8fb8852e1be23

not urgent, but because of the adafruit M4 boards, there is some demand.

ilg-ul, 2019-02-07:
thank you for the info, the next release will include it.

ilg-ul, 2019-04-29:
I just released 0.10.0-12, with the latest upstream master.

Please check if the problem was solved. If not, reopen the ticket.

coolio986, 2019-05-30:
I am running 20190426-0.10.0 trying to use atmel ice with SAMD51J19A and getting an error where it cannot find ID 60060005

> Open On-Chip Debugger 0.10.0 (2019-04-26) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> none separate
> adapter speed: 400 kHz
> cortex_m reset_config sysresetreq
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : CMSIS-DAP: SWD  Supported
> Info : CMSIS-DAP: JTAG Supported
> Info : CMSIS-DAP: FW Version = 01.27.0082
> Info : CMSIS-DAP: Serial# = J41800093192
> Info : CMSIS-DAP: Interface Initialised (SWD)
> Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
> Info : CMSIS-DAP: Interface ready
> Info : clock speed 400 kHz
> Info : SWD DPIDR 0x2ba01477
> Info : at91samd51j19a.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' connection on tcp/3333
> Error: Couldn't find part corresponding to DID 60060005
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected

openocd.cfg:

```
# Atmel-ICE JTAG/SWD in-circuit debugger.
interface cmsis-dap
cmsis_dap_vid_pid 0x03eb 0x2141

# Chip info 
set CHIPNAME at91samd51j19a
source [find target/at91samdXX.cfg]

cmsis_dap_serial J4180009XXXX
```

coolio986, 2019-05-30:
I got my dates wrong. your merge was 29th, my version is from days before

Do you have a link to the windows binary for the 29th?

ilg-ul, 2019-05-30:
I re-opened the ticket, but I have no clue what happens.

do you know of any openocd distribution that behaves as expected? if so, can you compare the configure options, and try to identify what is missing or is different?

ivankravets, 2019-05-30:
@coolio986 please use this binary https://github.com/gnu-mcu-eclipse/openocd/releases

coolio986, 2019-05-30:
Ok, I tested 20190422 with this result:

> GNU MCU Eclipse OpenOCD, 64-bitOpen On-Chip Debugger 0.10.0+dev-00593-g23ad80df4 (2019-04-22-20:25)
> Licensed under GNU GPL v2
> For bug reports, read
>         http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
> none separate
> adapter speed: 400 kHz
> cortex_m reset_config sysresetreq
> Info : Listening on port 6666 for tcl connections
> Info : Listening on port 4444 for telnet connections
> Info : CMSIS-DAP: SWD  Supported
> Info : CMSIS-DAP: JTAG Supported
> Info : CMSIS-DAP: FW Version = 01.27.0082
> Info : CMSIS-DAP: Serial# = J41800093192
> Info : CMSIS-DAP: Interface Initialised (SWD)
> Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
> Info : CMSIS-DAP: Interface ready
> Info : clock speed 400 kHz
> Info : SWD DPIDR 0x2ba01477
> Info : at91samd51j19a.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : accepting 'gdb' connection on tcp/3333
> Error: Couldn't find part corresponding to DID 60060005
> Error: auto_probe failed
> Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
> Error: attempted 'gdb' connection rejected

ilg-ul, 2019-09-24:
Please check the latest release, now part of the xPack project.

If you still encounter problems, report them upstream.

closed 2019-09-24 by ilg-ul