ilg-archived/openocd #38: Beaglebone Black : IR capture error

r00tb3, 2019-02-13:
I'm trying to JTAG BBB using J-Link adapter and I end up facing the following error:

`GNU MCU Eclipse 64-bit Open On-Chip Debugger 0.10.0+dev-00352-gaa6c7e9b (2018-10-20-06:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 800 kHz
trst_and_srst separate srst_gates_jtag trst_push_pull srst_open_drain connect_deassert_srst
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.410 V
Info : clock speed 800 kHz
Info : JTAG tap: am335x.jrc tap/device found: 0x2b94402f (mfg: 0x017 (Texas Instruments), part: 0xb944, ver: 0x2)
Info : JTAG tap: auto0.tap tap/device found: 0xffffffff (mfg: 0x7ff (<invalid>), part: 0xffff, ver: 0xf)
Error: auto0.tap: IR capture error; saw 0x0003 not 0x0001
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
Info : Listening on port 3334 for gdb connections
`
I'm unable to halt the processor after opening up a telent session nor able to do anything else:

`Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
> halt
Target not examined yet
`

These are my am335x.cfg file for BBB:

`source [find target/icepick.cfg]

if { [info exists CHIPNAME] } {
	set _CHIPNAME $CHIPNAME
} else {
	set _CHIPNAME am335x
}

# set the taps to be enabled by default. this can be overridden
# by setting DEFAULT_TAPS in a separate configuration file
# or directly on the command line.
if { [info exists DEFAULT_TAPS] } {
	set _DEFAULT_TAPS "$DEFAULT_TAPS"
} else {
	set _DEFAULT_TAPS "$_CHIPNAME.tap"
}

#
# Main DAP
#
if { [info exists DAP_TAPID] } {
	set _DAP_TAPID $DAP_TAPID
} else {
	set _DAP_TAPID 0x4b6b902f
}
jtag newtap $_CHIPNAME tap -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_DAP_TAPID -disable
jtag configure $_CHIPNAME.tap -event tap-enable "icepick_d_tapenable $_CHIPNAME.jrc 12 0"
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.tap

#
# M3 DAP
#
if { [info exists M3_DAP_TAPID] } {
	set _M3_DAP_TAPID $M3_DAP_TAPID
} else {
	set _M3_DAP_TAPID 0x4b6b902f
}
jtag newtap $_CHIPNAME m3_tap -irlen 4 -ircapture 0x1 -irmask 0xf -expected-id $_M3_DAP_TAPID -disable
jtag configure $_CHIPNAME.m3_tap -event tap-enable "icepick_d_tapenable $_CHIPNAME.jrc 11 0"
dap create $_CHIPNAME.m3_dap -chain-position $_CHIPNAME.m3_tap

#
# ICEpick-D (JTAG route controller)
#
if { [info exists JRC_TAPID] } {
	set _JRC_TAPID $JRC_TAPID
} else {
	set _JRC_TAPID 0x0b94402f
}
jtag newtap $_CHIPNAME jrc -irlen 6 -ircapture 0x1 -irmask 0x3f -expected-id $_JRC_TAPID -ignore-version
jtag configure $_CHIPNAME.jrc -event setup {
	global _DEFAULT_TAPS
	enable_default_taps $_DEFAULT_TAPS
}
# some TCK tycles are required to activate the DEBUG power domain
jtag configure $_CHIPNAME.jrc -event post-reset "runtest 100"
jtag newtap auto0 tap -irlen 2 -expected-id 0x00000000
#
# helper function that enables all taps passed as argument
#
proc enable_default_taps { taps } {
	foreach tap $taps {
		jtag tapenable $tap
	}
}

#
# Cortex-M3 target
#
set _TARGETNAME_2 $_CHIPNAME.m3
target create $_TARGETNAME_2 cortex_m -dap $_CHIPNAME.m3_dap

#
# Cortex-A8 target
#
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_a -dap $_CHIPNAME.dap -dbgbase 0x80001000

# SRAM: 64K at 0x4030.0000; use the first 16K
$_TARGETNAME configure -work-area-phys 0x40300000 -work-area-size 0x4000

# when putting the target into 'reset halt', we need to disable the watchdog as
# it would otherwise trigger while we're in JTAG
# FIXME: unify with target/am437x.cfg
source [find mem_helper.tcl]
set  WDT1_BASE_ADDR                  0x44e35000
set  WDT1_W_PEND_WSPR                [expr       $WDT1_BASE_ADDR     +  0x0034]
set  WDT1_WSPR                       [expr       $WDT1_BASE_ADDR     +  0x0048]
proc disable_watchdog { } {
	global WDT1_WSPR
	global WDT1_W_PEND_WSPR
	global _TARGETNAME

	set curstate [$_TARGETNAME curstate]

	if { [string compare $curstate halted] == 0 } {
		set WDT_DISABLE_SEQ1	0xaaaa
		set WDT_DISABLE_SEQ2	0x5555

		mww phys $WDT1_WSPR $WDT_DISABLE_SEQ1

		# Empty body to make sure this executes as fast as possible.
		# We don't want any delays here otherwise romcode might start
		# executing and end up changing state of certain IPs.
		while { [expr [mrw $WDT1_W_PEND_WSPR] & 0x10] } { }

		mww phys $WDT1_WSPR $WDT_DISABLE_SEQ2
		while { [expr [mrw $WDT1_W_PEND_WSPR] & 0x10] } { }
	}
}
$_TARGETNAME configure -event reset-end { disable_watchdog }
`
Where am I going wrong?

ilg-ul, 2019-02-13:
I don't know. 

Did you try the same sequence of commands with an OpenOCD build from the master and it works?

r00tb3, 2019-02-13:
@ilg-ul No but as you can see from above I'm using the 2018-10-20 version i think this will be stable.

ilg-ul, 2019-02-13:
> the 2018-10-20 version i think this will be stable

negative. those versions included some RISC-V code which affected the other targets.

not to mention there is no such thing as a stable OpenOCD version.

try the latest release.

anyway, before reporting problems here be sure they are specific to the GNU MCU Eclipse distribution, which means a similar version built from the master sources works.

if the problem also occurs in the master, please report it to https://sourceforge.net/p/openocd/tickets/

r00tb3, 2019-02-13:
@ilg-ul Okay!! I'll try the master version and the latest release let you know soon.

Thanks for the response.

r00tb3, 2019-02-14:
> negative. those versions included some RISC-V code which affected the other targets.
>
> not to mention there is no such thing as a stable OpenOCD version.
>
> try the latest release.
>
> anyway, before [...]

I tried as you said git cloned the master branch built it still gives me the same error.

I don't know where I'm going wrong.

ilg-ul, 2019-02-14:
sorry, I don't use openocd, and I don't have experience with the BB. :-(

closed 2019-02-14 by ilg-ul