ilg-archived/openocd #39: build against libusb 1.0 with usbdk support?

akohlsmith, 2019-04-16:
The binaries work great as long as you are using devices with the winusb/libusb driver installed against them; for some devices (such as FTDI devices) you can use Zadig to switch over to the winusb driver, but then anything that is using that FTDI device with the native drivers will no longer work.

It looks like libusb 1.0.21 and later support usbdk, which brings some sanity to the Windows world of USB; with usbdk support, libusb can request that Windows detach the driver so that it can be used with libusb directly, neatly solving the issue.

Is there any possibility of a binary build of openocd linked against libusb 1.0.21+ so we can take advantage of this ability?

ilg-ul, 2019-04-16:
I don't know if the OpenOCD configure script supports it. 

Is there any OpenOCD binary for Windows that uses this library? Could you try building OpenOCD from sources and, if it works, share the configuration?

ilg-ul, 2019-04-16:
I'm a bit confused by libusb distributions. For now I'm using the one from SourceForge (the next release planned soon uses https://sourceforge.net/projects/libusb/files/libusb-1.0/libusb-1.0.22/).

However I also noticed https://github.com/libusb/libusb, which seems more recent.

Do you know the differences? Which one do you recommend, considering the distribution should include binaries for all platforms?

ilg-ul, 2019-04-16:
Arch Linux also uses the SourceForge library for the mingw libusb package:

https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=mingw-w64-libusb

ilg-ul, 2019-04-22:
Could you try the 0.10.0-12 pre-release and confirm if the new libusb fixes the problem?

akohlsmith, 2019-04-24:
Thank you I will check this ASAP; that was *incredibly* fast, thank you!

ilg-ul, 2019-04-29:
I just released 0.10.0-12, with the latest upstream master.

Please check if the problem was solved. If not, reopen the ticket.

closed 2019-04-29 by ilg-ul

ivankravets, 2019-05-24:
We received some reports that openOCD has stopped to work with ST-Link probe on Windows. The previous version works without any issues. Is this problem linked with this issue?

ilg-ul, 2019-05-24:
my test include a debug session with ST-Link with a STM32F4DISCOVERY, and in my setup it was ok.

can you be more specific? also please note that I do not know the details of the windows usb drivers.

ivankravets, 2019-05-24:
> can you be more specific?

I will try to get more details from a user.