ilg-archived/openocd #42: OpenOCD fails to flash STM32H743xx rev V silicon

theNth, 2019-09-24:
### Description

OpenOCD successfully flashes STM32H743xx rev Y silicon, but fails to flash STM32H743xx rev V silicon. From the command line, running:
  `./src/openocd -s ./tcl -f ./tcl/board/st_nucleo_h743zi.cfg -c "program /path/to/file.elf verify reset exit"`
produces the following output when trying to program an STM32H743VIT6-V on a custom PCBA:

```Open On-Chip Debugger 0.10.0+dev-00594-g20b0eca04 (2019-09-23-16:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1800 kHz
adapter_nsrst_delay: 100
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : clock speed 1800 kHz
Info : STLINK V2J29S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.116230
Info : stm32h7x.cpu: hardware has 8 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0xfffffffe msp: 0xfffffffc
adapter speed: 4000 kHz
** Programming Started **
auto erase enabled
Info : Device: STM32H7xx 2M
Info : flash size probed value 2048
Info : STM32H flash has dual banks. Bank (0) size is 1024kb, base address is 0x8000000
Error: flash write algorithm aborted by target
Info : error executing stm32h7x flash write algorithm
Error: flash write failed, FLASH_SR = 00050000
Error: error writing to flash at address 0x08000000 at offset 0x00020000
** Programming Failed **
shutdown command invoked
```

Trying to flash an STM32H743ZIT6U-V on a NUCLEO-H743ZI2 dev board produces similar results. However, flashing an STM32H743ZIT6-Y on a NUCLEO-H743ZI succeeds.

### Steps to Reproduce

1. Attempt to flash an H743ZI or H743VI rev V part from a bash shell, e.g.:

```
./src/openocd -s ./tcl -f ./tcl/board/st_nucleo_h743zi.cfg -c "program /path/to/file.elf verify reset exit"
```

**Expected behaviour:** The binary should be successfully flashed onto the H7 with the following output:
```** Programming Started **
auto erase enabled
Info : Device: STM32H7xx 2M
Info : flash size probed value 2048
Info : STM32H flash has dual banks. Bank (0) size is 1024kb, base address is 0x8000000
wrote 131072 bytes from file /path/to/file.elf in 1.826331s (70.086 KiB/s)
** Programming Finished **
** Verify Started **
verified 14552 bytes in 0.103641s (137.117 KiB/s)
** Verified OK **
** Resetting Target **
shutdown command invoked
```

**Actual behaviour:** Flashing fails with the following output:
```** Programming Started **
auto erase enabled
Info : Device: STM32H7xx 2M
Info : flash size probed value 2048
Info : STM32H flash has dual banks. Bank (0) size is 1024kb, base address is 0x8000000
Error: flash write algorithm aborted by target
Info : error executing stm32h7x flash write algorithm
Error: flash write failed, FLASH_SR = 00050000
Error: error writing to flash at address 0x08000000 at offset 0x00020000
** Programming Failed **
shutdown command invoked
```

### Versions

* OpenOCD package version:
  0.10.0+dev-00594-g20b0eca04
  Built from 20b0eca0490fbc4f38f69eed8542cb082b354b03
* plug-in version: N/A
* Eclipse version: N/A
* Java version:
openjdk 11.0.3-internal 2019-04-16
OpenJDK Runtime Environment (build 11.0.3-internal+0-adhoc..jdk11u-jdk-11.0.3-ga)
OpenJDK 64-Bit Server VM (build 11.0.3-internal+0-adhoc..jdk11u-jdk-11.0.3-ga, mixed mode)
* operating system
Linux 4.15.0-60-generic #67-Ubuntu SMP x86_64 GNU/Linux

### Fix

The problem appears to be a consequence of the `stm32x_write` function in `src/flash/nor/stm32h7x.c` not setting the PGx bit in the bank x flash control register prior to calling `stm32x_write_block`. Inserting a call to `target_write_u32` anywhere before `target_run_flash_async_algorithm` to set PGx appears to fix the problem. Based of the existing code structure, this call should be made immediately prior to calling `stm32x_write_block`. See https://github.com/theNth/openocd/commit/972e7cd8c81fe501e3642bae8f0811ddb34d0686 for a proposed patch.

theNth, 2019-09-24:
The corresponding pull request may be found here: https://github.com/gnu-mcu-eclipse/openocd/pull/43

ilg-ul, 2019-09-24:
Thank you for your elaborate bug report and fix, but please address them upstream, this project only repacks the original sources, without adding functional changes.

closed 2019-09-24 by ilg-ul